jmsv/l293d #5: Needs option to use BCM GPIO numbering

jamesevickery, 2016-09-17:
`BOARD` numbers are currently used, but user could _also_ have the option to use `BCM` numbering.

See [pinout.xyz/](https://pinout.xyz/) for reference.

SmithcS, 2016-09-25:
I'm not sure if I can assign myself to this, but I'll start working on this (and possibly issue #10) too

jamesevickery, 2016-09-25:
@SmithcS Yeah it says you've assigned yourself :) Btw do you have Twitter or anything that I can IM you on? I can do #10 if you want, it's just a quick one: I wanted some of the code at the top of l293d.py to be put into methods.

SmithcS, 2016-09-26:
I do have an ancient Twitter account ill bring back to life - I pass you my handle after I reset the password

jamesevickery, 2016-09-28:
@SmithcS Okay thanks. Are you getting very far with this issue? Don't worry if not, I could do it soon

SmithcS, 2016-09-28:
Hey - yep I started working on it Monday, however I've been busy at work
the past couple days. Sorry for leaving it in limbo, I'll probably finish
up with it sometime today/tomorrow.

jamesevickery, 2016-09-30:
@SmithcS Could you get this checked in soon? There's some other stuff I want to do but it would be a nightmare trying to merge it 👍

SmithcS, 2016-10-01:
Hey! Sorry that took so long. Just made a pull request for you to check in.

jamesevickery, 2016-10-01:
Thanks! Btw I don't actually have L293D chips and motors with me for a while - if you've been using this library could you tell me if something stops working? I've only been developing while `test_mode = True`.

closed 2016-10-01 by jamesevickery

SmithcS, 2016-10-01:
I have some motors but not the chips, I'll order some off of amazon.