jmsv/l293d #20: Stepper motor support

jamesevickery, 2016-10-02:
This issue might be a long shot - I've currently never experimented with using the L293D for stepper motors, but I know that this chip does support them.

I'll buy a stepper motor at some point in the (distant) future and have a go at writing something to try and make it work. In the meantime, **if you stumble upon this and reckon you could have a go at it, I would be very grateful if you did** - just assign yourself and/or comment to say you're having a go.

After a quick Google I found [this Adafruit lesson](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-10-stepper-motors/overview) which looks like it might be a good place to start. [This tutorial](http://tutorials-raspberrypi.com/how-to-control-a-stepper-motor-with-raspberry-pi-and-l293d-uln2003a/) also looks helpful.

SmithcS, 2016-10-05:
I've done some stuff with stepper motors before. Kinda strapped on time, but I'll see if I can get this working. Also just got my IC's today so i'll be able to give it a test.

jamesevickery, 2016-10-05:
@SmithcS I've ordered a stepper motor online but it'll take weeks to come. If you were to add support for stepper motors could you also add some documentation in the README? Also, did you get your Twitter account sorted out? You can [DM me here](http://twitter.com/jamesevickery/)

For developing and documenting on this issue, the setup shown in the Adafruit lesson (link in issue description) looks like a good way of doing it.

As the `l293d.driver.motor` class has been developed for use with DC motors, it might be worth renaming `motor` to `dc_motor` or `dc`, and creating a new class called `stepper` or `stepper_motor` for use with stepper motors.

jamesevickery, 2016-11-01:
The `Motor` class has been renamed to '`DC`'. See commit referenced in issue #48.

jamesevickery, 2016-11-17:
@SmithcS Are you still interested in this issue? Don't worry if not - I can have a look at some point in the future