jmsv/l293d #23: Travis CI tests

jamesevickery, 2016-10-03:
Tests for this library need to be set up in `.travis.yml`. This issue is a sequel to issue #16.

surajnarwade, 2016-10-21:
which kind of test ?
can you please elaborate ?

jamesevickery, 2016-10-22:
@surajnarwade I'm not sure to be honest - I have no experience in unit testing Python so thought I should try and add some tests 😕. Feel free to suggest anything if you have any ideas!

surajnarwade, 2016-10-22:
I can add linting in travis for you, but can unit test utility also before that you have to write unit test for project

jamesevickery, 2016-10-22:
@surajnarwade Linting would be very helpful thanks, do you recommend any particular Python unit test method?

surajnarwade, 2016-10-23:
@jamesevickery , I am adding linting, go with simple unit testing

PR #34 (2016-10-23): Added Travis CI
Fixes #23
commit 80bd9d12 (2016-10-23): Added Travis CI

Fixes #23
closed 2016-10-23 by jamesevickery