jmsv/l293d #24: `setup.py` dependancies check

jamesevickery, 2016-10-03:
Before `setup.py` runs `install`, there should be a check to make sure the user has installed the dependencies. If they haven't, there can be an option to run `install` anyway, or abort the setup.

surajnarwade, 2016-10-21:
hello, can you provide llist of dependencies so that I can contribute here ?

jamesevickery, 2016-10-22:
@surajnarwade Thanks for the question, these should be installed:

- python-dev
- python-pip
- python-yaml

And there should be a warning if `RPi.GPIO` isn't installed or something, although if it's not installed then `driver.py` forces `test_mode = True`.

surajnarwade, 2016-10-22:
can you label this issue as hacktoberfest?

jamesevickery, 2016-10-22:
Reopening because I wanted `setup.py` to raise exceptions if importing the dependencies fails - but thanks for the change @surajnarwade!

jamesevickery, 2016-10-22:
`ImportError` raised if `import yaml` fails. Exception message prompts user to install `python-yaml`.

PR #32 (2016-10-22): Updated setup.py dependencies check
Fixes #24
commit 6ae3c95a (2016-10-22): Updated setup.py dependencies check

Fixes #24
closed 2016-10-22 by jamesevickery