jmsv/l293d #35: Fix linting warnings

jamesevickery, 2016-10-23:
@surajnarwade has added linting with flake8, and there are lots of warnings. I've started going through them by running `flake8` locally. Most warnings are for `driver.py`

jamesevickery, 2016-10-23:
I've tagged this as hacktoberfest because if you're looking for something easy to submit a PR for, this is it! I'll accept PRs that don't fix every single one, just mention this issue in the commit message (e.g. "Worked on #35")

vlvrd, 2016-10-23:
Hey, I wanna give it a go.

jamesevickery, 2016-10-23:
@vlvrd Feel free! Some basic instructions:

1. Install flake8 (pip install flake8)
2. Run `flake8`
3. Go through the list and make the suggested changes

jamesevickery, 2016-10-23:
Fixed by @PatOConnor43 in PR #37

closed 2016-10-23 by jamesevickery