jmsv/l293d #43: Travis CI failing with ImportError

jamesevickery, 2016-10-26:
[Build 64](https://travis-ci.org/jamesevickery/l293d/builds/170746840):

`ImportError: No module named l293d.driver`
...
`The command "python tests/l293d-tests.py" exited with 1.`

Not sure why this is happening, as `.travis.yml` runs `sudo python setup.py install` earlier in the script.

jamesevickery, 2016-10-26:
I'm very new to testing in Travis CI so help would be appreciated

lokeshh, 2016-10-26:
Is this build of the current master?

jamesevickery, 2016-10-26:
@lokeshh Yes

the-zebulan, 2016-10-26:
@jamesevickery ,

What version of Python are you using? The reason I ask is, you are using `print()` as a function and `import yaml` which only works (by default) with Python 3. Your `.travis.yml` file is using `Python 2.7` to run the test in `l293d-tests.py`.

I got the test to run using Python 3.5 with a small change to a couple import statements. I could send a pull request with the changes but I wasn't sure which version(s) of Python you wanted to use. The code is slightly different for using Python 2 vs Python 3.

jamesevickery, 2016-10-26:
@the-zebulan Python 2.7. The test passes locally on 2.7, but I think '`ImportError: No module named l293d.driver`' means that there's a problem installing, or the install directory isn't in the Python path.

the-zebulan, 2016-10-26:
@jamesevickery ,

Oh ok, I made an incorrect assumption that you were using Python 3. I'll switch my interpreter to Python 2 to test this out.

If you want to use `print()` with Python 2, you could use `from __future__ import print_function` and get all the extra benefits too!

jamesevickery, 2016-10-26:
@the-zebulan 
I've been looking into the `print` differences between Python 2 and 3 - I made a Python file:
`import sys`
`print(sys.version)`

 and ran it using `python filename.py`,
the output was:
`2.7.9 (default, Sep 17 2016, 20:26:04)`
`[GCC 4.9.2]`

the-zebulan, 2016-10-26:
Without the import from `__future__`, you can remove the brackets and it shouldn't change the output.

Try typing this with Python 2:

```
>>> import sys
>>> print(sys.version, end='$$$')
  File "<stdin>", line 1
    print(sys.version, end='$$$')
                          ^
SyntaxError: invalid syntax
```

jamesevickery, 2016-10-26:
@the-zebulan Yeah I see - so is there anything wrong with using `print('')` in Python 2? Because I am at the moment and it's been working.

Back to the issue though, I want to use Python 2 (although I'm considering adding support for Python 3 in the future)

the-zebulan, 2016-10-26:
@jamesevickery ,

It can confuse other people about which version of Python you are using (like me!). You just don't get to use any of the extra functionality that comes with having `print` be a function. One of the first steps to having a cross-compatible script (if you needed `print` in both) would be adding `from __future__ import print_function`.

As you said, back to the issue. Do you get these two `RuntimeWarning`'s when you run your test file?

```
l293d/tests/l293d-tests.py:1: RuntimeWarning: Parent module 'tests' not found while handling absolute import
  import unittest
l293d/tests/l293d-tests.py:7: RuntimeWarning: Parent module 'tests' not found while handling absolute import
  import l293d_chip.driver
```

jamesevickery, 2016-10-26:
@the-zebulan I have another issue open for future Python 3 support if you want to add stuff there, thanks for the `print` clarification.

No, my test works locally but not on Travis CI. **Local output**:
```
L293D driver version 0.2.0
Can't import RPi.GPIO. Please (re)install.
Test mode has been enabled. Please view README for more info.
<module 'l293d.driver' from '/usr/local/lib/python2.7/dist-packages/l293d/driver.pyc'>
.
----------------------------------------------------------------------
Ran 1 test in 0.027s

OK
```

**Travis CI output**:

```
E
======================================================================
ERROR: test_import (__main__.tests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "tests/l293d-tests.py", line 7, in test_import
    import l293d.driver
ImportError: No module named l293d.driver
----------------------------------------------------------------------
Ran 1 test in 0.000s
FAILED (errors=1)
```

PR #44 (2016-10-26): update_for_travis
I think this PR will close #43 . I had to try a few different ideas before I could get it to work.

* I added an `__init__.py` file to the `tests` directory (for the `ImportError`)
* I updated the `.travis.yml` file
    * I used `py.test` as the testrunner (only because I see it recommended lots and I have used it before too)
    * `pip` is already installed (as per the Travis docs) so I removed the install `pip` part
    * I was having trouble with `python-yaml` install but replacing it with `pip install pyyaml` seems to work fine
* I renamed the test file to start with `tes [... cut]
closed 2016-10-26 by jamesevickery