jmsv/l293d #54: Installation directory config

jamesevickery, 2016-10-30:
Idea from PR #53 comments:

> - The 'default' `.ini` config stays where it is without being copied anywhere
- New methods are made in `config.py` to let the user modify the parameters' values without having to manually edit the file.
  - For example, the user is prompted for values for each parameter (or they can hit return to leave them the same) and then the new config is written to the file
- When `config.py` is run by `driver.py`, config is loaded from the file in the installation directory.

> I like the idea of the config file being in the `l293d` installation directory but editable by the user - it keeps everything in one place and keeps the user's home tidy.

the-zebulan, 2016-10-30:
@jamesevickery ,

Do you mean something like this:

```python
class L293DConfig(object):
    def __init__(self,
                 config_filename='l293d_config.ini',
                 pin_numbering='BOARD',
                 test_mode=False,
                 verbose=True):

# and then when you use it you can set the parameters when creating the instance?
config = L293DConfig(pin_numbering='BCM', test_mode=True, verbose=False)
```

jamesevickery, 2016-10-31:
@the-zebulan I'm not sure I explained it very well - I'll have a go at implementing it and hopefully you should see what I mean

jamesevickery, 2017-05-24:
Closed, see new issue #58

closed 2017-05-24 by jamesevickery