jmsv/l293d #56: PWM support

icapan, 2017-04-18:
Great stuff. I found out about your project following a link from the original at the tutsplus.com. One thing I would find useful: PWM. Controlling the speed of the motors. Check out [this link](https://raspberrypi.stackexchange.com/questions/56718/pwm-dc-motor-using-l293d-weak-sound)

jamesevickery, 2017-04-18:
@icapan Thanks for the interest! I was considering adding support for PWM ages ago but never got around to it - there are a few things (e.g. #20) that I would like to add but I'm currently spending most of my time on other projects.

If you've got an L293D chip set up at the moment and feel like contributing, feel free to have a go at adding PWM support yourself! I'd be happy to get back into this project and help

icapan, 2017-04-19:
I have a chip and several DC motors, I intend to try to configure it (I wasn't successful so far), but my programming knowledge is limited, so don't expect too much from me :)

jamesevickery, 2017-04-19:
@icapan That's alright, just feel free to fork this repo and submit a pull request if you get anywhere

jamesevickery, 2017-06-02:
Some useful links:
- http://raspi.tv/2013/rpi-gpio-0-5-2a-now-has-software-pwm-how-to-use-it
- https://pypi.python.org/pypi/RPi.GPIO
- https://www.coursera.org/learn/raspberry-pi-platform/lecture/DMArS/lecture-2-2-pulse-width-modulation

jamesevickery, 2018-01-22:
@icapan If you still need PWM support, @alxwrd has added it! I've bumped the version to 0.3.0 and this is on [PyPI](https://pypi.python.org/pypi/l293d/0.3.0), so `pip install l293d` should get the version with PWM support.

alxwrd, 2018-02-06:
Just a mention for @mortenmoulder as they were interested in PWM support from #61.

closed 2018-03-09 by jamesevickery