jmsv/l293d #57: Attribute errors

monoxane, 2017-05-22:
My code just will not run, I keep getting attribute errors as follows:

```
L293D driver version 0.2.2
Setting GPIO mode: BOARD
Traceback (most recent call last):
  File "forklift.py", line 35, in <module>
    forkliftSetup()
  File "forklift.py", line 31, in forkliftSetup
    driveLeft = l293d.Motor(22, 18, 16)
AttributeError: 'module' object has no attribute 'Motor'
```

```
L293D driver version 0.2.2
Setting GPIO mode: BOARD
Traceback (most recent call last):
  File "forklift.py", line 35, in <module>
    forkliftSetup()
  File "forklift.py", line 31, in forkliftSetup
    driveLeft = l293d.motor(22, 18, 16)
AttributeError: 'module' object has no attribute 'motor'
```

```
L293D driver version 0.2.2
Setting GPIO mode: BOARD
Traceback (most recent call last):
  File "forklift.py", line 35, in <module>
    forkliftSetup()
  File "forklift.py", line 31, in forkliftSetup
    driveLeft = l293d.DC(22, 18, 16)
AttributeError: 'module' object has no attribute 'DC'
```

What am I doing wrong?

jamesevickery, 2017-05-22:
What's your import statement? `DC` is part of `driver` which is part of `l293d`, so you'd need to use `import l293d.driver as l293d`

monoxane, 2017-05-22:
Ah I was just doing 
`import l293d.driver`
Looks like I didn't read everything, sorry for wasting your time.

closed 2017-05-22 by monoxane

jamesevickery, 2017-05-22:
@monoxane Don't worry about it - no time wasted. Admittedly it's not very intuitive, I might consider changing it to just `import l293d` in the future. I'll try and remember to reference this issue if I do.

jamesevickery, 2017-05-24:
@monoxane You can use `import l293d` now once you've updated to the latest version (0.2.3) and then things like `driveLeft = l293d.DC(22, 18, 16)` will work

monoxane, 2017-05-25:
Thank you very much.

jamesevickery, 2017-10-12:
@monoxane I just had a look at your forklift blog post - anything I can do to help?