jmsv/l293d #61: Cannot use two motors at once

mortenmoulder, 2017-06-01:
Hi, James! I absolutely love this project and I think it needs more attention. However, I have a weird issue: I cannot use two motors at once. The console outputs the same motor twice, but I know 100% sure I am using two different variables. Here's what my code looks like:

    import l293d

    motor1 = l293d.DC(15, 13, 11)
    motor2 = l293d.DC(22, 18, 16)

    motor1.clockwise()
    motor2.clockwise()

I get this console output:

    pi@raspberrypi:~ $ python3 motor.py
    L293D driver version 0.2.4
    Setting GPIO mode: BOARD
    spinning motor at BOARD pins [22, 18 and 16] clockwise.
    spinning motor at BOARD pins [22, 18 and 16] clockwise.

But you can clearly see I use different pins and different variables. Do you know why this is happening?

jamesevickery, 2017-06-01:
Oh dear that doesn't look good - that was probably introduced when I changed how the package is imported. I'll make sure I can reproduce this and hopefully be able to fix this within the next hour.

Thanks so much for raising an issue and not abandoning the project! Raising this was extremely helpful and is more evidence that my tests need to be a lot better.

mortenmoulder, 2017-06-01:
No problem! I prefer working with this over setting everything to HIGH or LOW all the time haha.

Thanks a lot!

jamesevickery, 2017-06-01:
That should of fixed it (ce9adc0)

I'll push to PyPI in a second - your code should work after running `pip install l293d --upgrade`

closed 2017-06-01 by jamesevickery

mortenmoulder, 2017-06-01:
Works perfectly! Thanks a lot!

jamesevickery, 2017-06-01:
Latest version is now on PyPI - I kept noticing things wrong so I ended up bumping the version a few times from 0.2.4 to 0.2.7

jamesevickery, 2017-06-01:
Thanks again for opening this issue - this bug has been lingering for longer than I had originally thought

mortenmoulder, 2017-06-02:
No worries, @jamesevickery! I got it all working and I think it's pretty good now!

PWM support would definitely be awesome to have as well, but I guess we can't have it all haha.

jamesevickery, 2017-06-02:
@mortenmoulder Thanks! If you want to contribute to PWM at all there's an issue at #56 - any ideas would be helpful and a code contribution would be even more so!

mortenmoulder, 2017-06-02:
Yeah I noticed that yesterday. If I had more knowledge with PWM and GPIO, I would give it a go, but unfortunately it's a bit out of my league right now 😞

jamesevickery, 2017-06-02:
@mortenmoulder Don't worry :grinning: I didn't really know anything about GPIO but just read lots of stuff online and experimented a bit - I've put a list of a few links that look helpful in the comments for #56, feel free to have a look. I'll have more of a look at PWM hopefully over the next few weeks.

commit ce9adc04 (2017-06-01): Fixed #61