jmsv/l293d #62: Add tests to avoid problems like #61

jamesevickery, 2017-06-02:
Test coverage is currently awful - many more need to be made to prevent basic problems like #61

alxwrd, 2018-02-06:
hi @jamesevickery! I'm hoping to take a look at adding some tests this week/weekend. Have you had any thoughts on this? i.e. any weird edge cases?

A couple of links about mocking the RPi.GPIO module:

- https://stackoverflow.com/questions/32508349
- https://www.reddit.com/r/Python/comments/5eddp5

Mock objects look really useful for testing functionality around GPIO:

```python
>>> from unittest.mock import MagicMock
>>> GPIO = MagicMock()

>>> GPIO.setmode(GPIO.BOARD)
<MagicMock name='mock.setmode()' id='140549155481584'>

>>> GPIO.setmode.assert_called_with(GPIO.BOARD)

>>> GPIO.setmode.assert_called_with(GPIO.BCM)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/unittest/mock.py", line 814, in assert_called_with
    raise AssertionError(_error_message()) from cause
AssertionError: Expected call: setmode(<MagicMock name='mock.BCM' id='140549155494208'>)
Actual call: setmode(<MagicMock name='mock.BOARD' id='140549155469016'>)
```

jamesevickery, 2018-02-09:
@alxwrd This looks good 😄

Using `MagicMock` sounds like a good idea, feel free to add tests