jmsv/l293d #72: l293d output should be annotated

jamesevickery, 2017-07-04:
When the library prints something, it should be prefixed by '`[l293d] `'

crispg72, 2017-10-12:
Hi @jamesevickery would this involve updating [this](https://github.com/jamesevickery/l293d/blob/master/l293d/driver.py#L11-L20) function to add that prefix?

Would you like the code refactored to use python's `logging` lib? That would make these kind of changes easier and also mean any exceptions would be logged in the same way.

I would be happy to create a pr #hacktoberfest

jamesevickery, 2017-10-12:
@crispg72 Yes, that's the right method.

I haven't used the `logging` lib before, although it looks more suited to what I'm using `print` for - feel free to make those changes. A PR would be perfect thanks

crispg72, 2017-10-13:
Ok great, I'm hoping to get this done later today.

crispg72, 2017-10-14:
Created https://github.com/jamesevickery/l293d/issues/72

closed 2017-10-16 by jamesevickery