jmsv/l293d #90: Problem with DC.stop

dgmp88, 2018-07-03:
Thanks for the cool library! 

I currently get an error whenever DC.stop is called. e.g. 

```python
import l293d
left = l293d.DC(22,18,16)
left.clockwise(duration=0.1, speed=100, wait=True)
l293d.cleanup()
```

gives 

> Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/lib/l293d/l293d/driver.py", line 160, in stop
    self.__move_motor(0, after, True, 'stopping', None)
  File "/home/pi/lib/l293d/l293d/driver.py", line 140, in __move_motor
    wait=wait, speed=speed)
  File "/home/pi/lib/l293d/l293d/driver.py", line 95, in drive_motor
    speed = PWM(*speed)
TypeError: type object argument after * must be an iterable, not NoneType

It still stops though. 

I'm happy to debug this and make a PR, but just wanted to check before I do if I'm doing anything wrong?

jmsv, 2018-07-03:
Oops :grimacing: 

Thanks for raising this! I've pushed a fix and I'll update the PyPI package now

closed 2018-07-03 by jmsv

jmsv, 2018-07-03:
Done - https://pypi.org/project/l293d/0.3.3

Let me know if this fixes your problem :smile:

dgmp88, 2018-07-03:
Insanely fast fix! Works for me now. Cheers 😁

commit 2a597222 (2018-07-03): Fixed #90