jmsv/l293d #94: Python Error when call to motor.stop() before call to motor.clockwise()

twinstef, 2020-04-15:
Code:

import l293d
door = l293d.DC(22, 18, 16)
door.stop()

Result:
  File "/usr/local/lib/python2.7/dist-packages/l293d/driver.py", line 162, in stop
    self.__move_motor(0, after, True, 'stopping', None)
  File "/usr/local/lib/python2.7/dist-packages/l293d/driver.py", line 142, in __move_motor
    wait=wait, speed=speed)
  File "/usr/local/lib/python2.7/dist-packages/l293d/driver.py", line 103, in drive_motor
    self.pwm.stop()
AttributeError: 'NoneType' object has no attribute 'stop'

jmsv, 2020-04-21:
thanks for raising this issue - pretty busy at the moment but I've put it on a todo list and will hopefully get to it at some point

twinstef, 2020-04-21:
not urgent, and thank's for your code which runs morning and evening in 
my chicken coop avoiding me to get up at dawn.

Stéphane

jmsv, 2020-04-21:
excellent to hear this is being put to good use! I would love to know more about your setup 😍